keepkey/keepkey-firmware #6: Ethereum Support

ming08108, 2016-05-18:
Now that trezor is planning to add ethereum support is there any plans for keepkey to add it also?

Since keepkey uses a fork of the firmware it should be relatively easy.

https://github.com/trezor/trezor-common/pull/11

Arachnid, 2016-05-18:
This recent activity was actually prompted by my own work on Ethereum support for Keepkey; see https://github.com/keepkey/keepkey-firmware/pull/5 and https://github.com/keepkey/device-protocol/pull/3.

I don't know how much @axic is planning to contribute upstream to Trezor; the current situation, with PRs flying about contributing modified or alternate versions of code is rather confusing, and so I'm going to halt my own work temporarily until I can be sure I won't be spending most of my time refactoring existing work.

prusnak, 2016-05-18:
@Arachnid As TREZOR is upstream it makes sense to contribute there first and let the KK guys do the rest of the work.

Arachnid, 2016-05-18:
I agree in principle, but that doesn't make things any easier from where I stand. I don't have a Trezor to test code on, so I can't contribute directly, but writing code for KK, then having it rewritten and submitted upstream, then having to handle merge conflicts with my original code(!) is just about the worst possible workflow.

prusnak, 2016-05-18:
We'll sort that out. :-)

bgok, 2016-05-18:
KeepKey is on board with this approach. @prusnak, are you planning to provide @Arachnid with a device to work with? If there is anything we can do to help move this along, let us know.

prusnak, 2016-05-20:
Yes, already sent.

bgok, 2016-05-20:
👍
Let me know if you need anything from me.

closed 2016-05-20 by bgok