keepkey/keepkey-firmware #17: How to prove signed firmware from KeepKey?

Aragonite, 2017-07-07:
Trevor has a process that anyone can follow to prove that their signed firmware binary was made of the github source code at a specific tag.   I don't see the tools available with KeepKey to prove the same.

For example, with Trezor, using the script firmware-fingerprint.sh, will remove the first 256 header bits from their official signed binary (where their signature is located) and a SHA256SUM of the rest of the file should match the SHA256SUM of the binary that can be compiled independently with the Github source code (using their script firmware-docker-build.sh).

The problem I am having with KeepKey, is their docker_build_release.sh script produces a binary file that has no header (the first 256 bits with the magic code KPKY is missing) - therefore, that binary file can't even be loaded into a KeepKey unit.  

Question 1:  How can we "sign" a firmware that we have compiled ourselves and load it to a KeepKey unit - with the understanding that unit will warn us every time we turn on the device that the firmware is not signed by KeepKey.

Question 2:  How can we verify that the binaries provided by KeepKey matches the code on Github, for each version?

yate, 2018-01-13:
For question 2, I believe it's a similar process to trezor. 

- Download the signed binary from a keepkey release and run `tail -c +257 keepkey_main.bin | sha256sum`
- Download the source code for the release version and run `./docker_build_release.sh`. The script outputs the sha256sum of the built firmware which should match with the above.

I'm interesting in the question 1 as well. I would like to have a developer guide on how to push your built firmware to your device and how to debug your code.

keepkeyjon, 2018-03-05:
For question 1:

You'll need to arrange for this metadata to be prepended to your binary (in order):

  * 4 bytes of magic: `"KPKY"`.
  * 4 bytes of firmware payload length, encoded as a little-endian uint.
  * 3 bytes of signature indexes. I suggest just using: `0x01, 0x02, 0x03`.
  * 1 byte of flags: `0x01`
  * 52 reserved bytes, all `0x00`
  * 3x bogus signatures, 64 bytes each. I believe this can either be all `0x00`s, or any random garbage. Your pick.
  * Followed by the firmware payload itself (i.e. `keepkey_main.bin`).

After "signing", you can upload with:

```
$ keepkeyctl wipe_device
$ keepkeyctl firmware_update --file path/to/bin/debug/keepkey/keepkey_main.bin
```

For question 2: @yate's suggestion should work. I believe you'll need to do a build with the fix in https://github.com/keepkey/keepkey-firmware/commit/c6c2c43627aa922d0506861ee3a5e02fd458b082, otherwise you might not get nice reproducible results that are consistent with the way we've built previous releases.

@yate w.r.t. debugging: it's a bit tricky, since we permanently disable jtag / on-chip debug / etc in production devices. Your best bet is probably to build a "debug link" firmware, and use `dbg_print()` to do printf debug over usb.

closed 2018-03-05 by keepkeyjon

keepkeyjon, 2018-03-05:
The structure of that metadata format is more precisely defined here, btw: https://github.com/keepkey/keepkey-firmware/blob/c6c2c43627aa922d0506861ee3a5e02fd458b082/keepkey_board/public/memory.h#L149