keepkey/keepkey-firmware #38: Process to add new coins/tokens

jleni, 2018-01-22:
The process to add support for new coins/tokens does not seem to be documented.

Are there any guidelines in place?

Positivism, 2018-06-29:
Modify then build the firmware.

keepkeyjon, 2018-09-05:
Our team is happy to review patches, and work with the community to add support for more coins. The general process would be to add support to `keepkey/device-protocol`, `keepkey/keepkey-firmware`, and `keepkey/python-keepkey`. Make sure all the tests still pass on `keepkey/python-keepkey`, and then submit PRs. We'll review them, and give feedback where appropriate.

On the client side, we'll need pointers to the appropriate Insight fork (or otherwise), and some time to stand up a node, as well as support in `keepkey/device-client`.

closed 2018-09-05 by keepkeyjon