keepkey/keepkey-firmware #45: Keepkey wallet (not dependent on chrome)

briannyeko, 2018-03-05:
when will you create a desktop app for the keepkey wallet(not dependent of chrome due to chrome apps only being restricted to chromebooks)?

keepkey-support, 2018-03-05:
Google has plans to discontinue Chrome app support and our development team is working very hard to ensure that a seamless solution is released prior to the Chrome app ending. This being said, we hope to have native support in the near feature! Keep and eye out for upcoming updates and supported features!

If you have anymore questions or concerns please reach out to support@keepkey.com

Thank you, 

-Andrew

_Certified Bitcoin Professional_

Support: help.keepkey.com
Website: www.keepkey.com

closed 2018-03-05 by keepkeyjon

MelbourneDeveloper, 2018-09-23:
@keepkey-support which platform will you be building the new solution on? Has development already started?

Many companies are choosing React Native, but in my opinion, Xamarin is a much better path to go down. Can you tell us anything about how the new app is shaping up?

keepkeyjon, 2019-03-16:
We've built a new integrated platform for ShapeShift + KeepKey, and are inviting folks to sign up for a closed beta here: https://shapeshift.io/wherecryptolives

MelbourneDeveloper, 2019-03-17:
Thanks for keeping me posted.

I’ve been intending on integrating Shapeshift, but I’ve just never gotten around to it. Keeping up with updates to all the devices, coins, and so on has kept my hands tied up. It’s time to open source the rest of my app. Once I’ve done that, I’m hoping to get a bit more help so I’m freed up to work on features.

I will check this platform out.