keepkey/keepkey-firmware #51: SSH Support Broken (Again)

NTICompass, 2018-07-19:
SSH support on the KeepKey was fixed in firmware 5.4.1, but now on the latest (5.6.1) and the new bootloader, SSH is no longer working.

In Linux, I am seeing the following with `keepkey-agent`:

```
Traceback (most recent call last):
  File "/home/nticompass/.local/lib64/python3.5/site-packages/libagent/device/trezor.py", line 151, in sign
    ecdsa_curve_name=curve_name)
  File "/home/nticompass/.local/lib64/python3.5/site-packages/keepkeylib/client.py", line 135, in wrapped_f
    ret = f(*args, **kwargs)
  File "/home/nticompass/.local/lib64/python3.5/site-packages/keepkeylib/client.py", line 690, in sign_identity
    return self.call(proto.SignIdentity(identity=identity, challenge_hidden=challenge_hidden, challenge_visual=challenge_visual, ecdsa_curve_name=ecdsa_curve_name))
  File "/home/nticompass/.local/lib64/python3.5/site-packages/keepkeylib/client.py", line 148, in wrapped_f
    return f(*args, **kwargs)
  File "/home/nticompass/.local/lib64/python3.5/site-packages/keepkeylib/client.py", line 189, in call
    msg = handler(resp)
  File "/home/nticompass/.local/lib64/python3.5/site-packages/keepkeylib/client.py", line 201, in callback_Failure
    raise CallException(msg.code, msg.message)
keepkeylib.client.CallException: (1, 'Unknown message')
```

NTICompass, 2018-07-30:
@keepkeyjon It looks like SSH support didn't make its way into the new cmake version of the firmware.  Can this be added back?  I've been using the KeepKey to SSH into things :-)

keepkeyjon, 2018-07-30:
Oops. We should have that fixed in the next firmware release.

closed 2018-07-30 by keepkeyjon

keepkeyjon, 2018-08-15:
[v5.7.0](https://github.com/keepkey/keepkey-firmware/releases/tag/v5.7.0), which is currently released in the [beta client](https://chrome.google.com/webstore/detail/keepkey-client-rc/ndpjhnboalogfiafbknodggihfiedibo) should fix this. Please let me know if you're still having issues with it.

NTICompass, 2018-08-16:
@keepkeyjon I updated to v5.7.0 and SSH is working again!  Thanks a lot, man :-)