keepkey/keepkey-firmware #55: Supporting all SIGHASH_ types

kyuupichan, 2018-10-27:
It would be nice to be able to support SIGHASH_SINGLE and SIGHASH_ANYONECANPAY; perhaps also SIGHASH_NONE, transaction signing.  The UI would need to warn / explain in some way.

keepkeyjon, 2018-10-29:
Yeah, that would be interesting. A bit difficult to explain to users in the general case though. SIGHASH_NONE in particular is a very sharp knife.

This might be better motivated with higher level on-device abstractions, i.e: `"Would you like to commit 0.001 BTC from Bitcoin account #0 to the 0.05 BTC fundraiser at 1bitcoineateraddressdontsendf59kue?"`.

Can you think of any other wallets that expose these hash types?

kyuupichan, 2018-10-29:
Not at present.  But I want to write something that would use them all other than SIGHASH_NONE

mrnerdhair, 2022-03-16:
I'm going to close this for now because of staleness, but also because we can't move forward until we have a template for specific language (or, more generally, UX) that would make sense for each sighash type. (Also, I expect that it would make the most sense to treat support for each of them as separate issues because developing that language will be a distinct process for each one.)

closed 2022-03-16 by mrnerdhair