keepkey/keepkey-firmware #75: Keepkey and BSV

kyuupichan, 2019-01-12:
Keepkey supports BSV transaction signing just fine by passing BitcoinCash as the coin name, and we use this in ElectrumSV.

However, for displaying bitcoin addresses for a user to confirm using client.get_address(), I cannot pass BitcoinCash as the coin name as we use original bitcoin addresses not cashaddr.  So I pass Bitcoin.  But then I get a nasty warning message about the address derivation (using BitcoinCash's 145' path) no being right because it expects 0'.

Could you add support for BSV as a separate coin type (with signing identical to BitcoinCash) that accepts BOTH 0' and 145' address derivations?  Because of history, we use both in BSV (and no others).

keepkeyjon, 2019-01-12:
@kyuupichan How do you feel about adding two coin table rows for BSV, one with `0'` as the slip44, and the other with `145'` respectively named `"BitcoinCashSV"` and `" BitcoinCashSV"`, so that `coinByName()` still works, but all the places we show it on the screen still look ok?

keepkeyjon, 2019-01-13:
Actually, thinking about this some more... the whole point of that warning is to encourage 3rd party wallets to use the correct SLIP44 derivation paths, and discourage other coins from using user's Bitcoin signing keys.

keepkeyjon, 2019-01-13:
BSV's SLIP44: https://github.com/satoshilabs/slips/pull/451

kyuupichan, 2019-01-13:
No wallet I know of, certainly not ElectrumSV, will use a new SLIP44 - it's a disaster for usability.  Many wallets seem to use 0 like Bitcoin; we have stuck with 145 but may also revert to 0.

keepkeyjon, 2019-01-13:
Using 0 will be pretty bad for privacy and security. The whole point of BIP44 separating coins with hardened derivation was to protect them from each other, so that a problem in the signing of one coin implementation does not spill over to a user's coins on another chain. I'm happy to add BSV support, but I don't want to do so in a way that puts our users at risk... In my book, usability always comes second to security.

closed 2019-01-13 by keepkeyjon

kyuupichan, 2019-01-13:
I'm aware of that.  However, keep in mind most users are new.  Those around at the split 18 months ago will have very likely already separated their coins and have them in different wallets

kyuupichan, 2019-01-15:
@keepkeyjon After discussions with fellow devs, let's go with this.

However you have made several changes that seem wrong.  Please copy the line for BCH and change only *2* fields:  BCH to BSV, and the account path.  You've changed several others without explanation (such as message signing) which will be incompatible with everyone else.

keepkeyjon, 2019-01-15:
I was under the impression that BSV folks were abandoning cashaddr.

kyuupichan, 2019-01-15:
Sorry yes, that's the 3rd item then!

keepkeyjon, 2019-01-15:
Is there new p2pkh/p2sh magic bytes? I saw an explorer with them, but can't find it now.

kyuupichan, 2019-01-15:
No.  Apart from cashaddr everything is the same.  And now the SLIP account derivation of course.

keepkeyjon, 2019-01-15:
https://github.com/keepkey/keepkey-firmware/pull/76/commits/4fef2af24f3900c0ecaedea3b60e56f6ef296554