keepkey/keepkey-firmware #85: Feature request: configurable discovery gap for small businesses.

nutmix, 2019-02-13:
I am a consultant for a number of businesses which accept payment, and make payments in crypto.

As a cryptocurrency developer, i am fully aware of the architecture, features and limitations of deterministic wallets.  On the plus side, you can generate, regenerate and recover any number of (indexed) addresses for any number of paths.  However, in practice, HW wallet developers have crippled this functionality by implementing a fixed discovery gap of 1, rendering HW wallets useless for anyone other than a single person with a single wallet.

Small Businesses, sole traders and consultants with multiple customers have the following requirements:

1. They need to assign an account (or at least an address) for each customer so that payments can be made in parallel.  Having one account for multiple customers to pay into violates requirements for customer fund segregation, accounting and audit.
1. They can't wait for customer A to pay his bill before they create an account/address for customer B. 
1. They need to generate many accounts/addresses up front (e.g. 10 or 100 at a time).
1. Ideally, they need to have an API to generate these addresses and accounts. 
1. Recovery (e.g. using same seed on a different HW dongle if first fails) does not need to be fully automated. E.g. it's perfectly acceptable to set the discovery gap limit if its not stored anywhere.
1. It doesn't matter if this device is a bit slow as it has to load 100 accounts.  They can live with that, but it should also be very easy to optimise this (e.g. local cache of accounts with a transaction + async loading)

Each business is currently using either Trezor or Ledger HW dongles, neither of which support multiple accounts.  Ledger is the most limiting.  Here is what two companies I consult for do:

Every day, they pay someone to do the following:
For each currency (they support 20+)

1. Check how many free (unassigned to a customer) addresses they have.
2. If < 5, they create a new address.
3. Then they transfer 0.00000001 into it (or equivalent)
4. They wait 1 hour for the transaction to complete.
5. They transfer the 0.00000001 out.
6. Now they are "allowed" to create a new address in the wallet SW.

I.e. they employ someone just to fund and create new addresses. When I showed this company a command line call to generate 100 addresses via the trezor api, they were blown away. They had no idea that a wallet could generate multiple addresses without putting the equivalent of 1p into them manually first.  Unfortunately, even though the trezor can generate unlimited addresses, it still wont "recognise" them unless you fund them.

I am currently advising muy clients to use the Electrum wallet + trezor where possible, as with the Electrum wallet you can set the discovery gap limit e.g to 20, and thus create 20 address in seconds (which again blew away these companies). However this only works for a limited set of tokens.

There are many simple to implement and easy for customers to use ways to handle the discovery gap, e.g:

1. Make the discovery gap limit configurable like in Electrum.  This is easy to use, easy to setup, easy to recover.
1. Store the discovery gap limit on a shared drive (like Trezor does for the account names).
1. store the highest index+ paths on a shared drive (like Trezor does for the account names)
1. Allow manually setting the highest index for easy recovery.
1. Allow import/ export of paths/indexes for easy recovery.
1. Simply allow addresses from any paths/indexes to be added manually.  Ideally allow export of list of addresses (and names if supported), their paths and indexes to spreadsheet.

keepkeyjon, 2019-02-13:
Thanks for the feedback!

Now that you mention it, I believe the KeepKey Client incorrectly sets the gap limit to 10. We ought to make that more standards compliant and change that to 20. As a general principle, we try follow the standardized address/account gap limits as a way to help users 'Do the RIght Thing' so as to maintain best portability of seed phrases into other wallets. While the features you're suggesting are useful, I don't see them as being appropriate for the more consumer-ish customers. Perhaps this means there's significant demand for a more... uh... retail... focused client offering. We'll certainly be considering that for the next revision of our frontend.

> I believe that the crippling limitations of the current HW wallet offerings are due to lack of innovation and understanding of basic accounting. Quoting BIPxxx as a justification for the limitations is just lazy and dangerous

Please table the insults. That's not an effective way to get what you want here.

closed 2019-02-13 by keepkeyjon

nutmix, 2019-02-13:
Sorry, this is not something that you have said. It is something that Ledger told their clients when they were asking for these features.   

You could consider either making these features in an "advanced" section, or add $200 to the price and sell it as a business solution - either way works.