keepkey/keepkey-firmware #86: Support Bitcoin Regtest

tynes, 2019-02-16:
It would be very useful for local development to have bitcoin regtest support in the coin table.

Here are the constants for regtest:
https://github.com/bcoin-org/bcoin/blob/f585d8616df8335fddbc21d1b776a979ec6ff305/lib/protocol/networks.js#L847

Satoshi Labs also has this slip:
https://github.com/satoshilabs/slips/blob/master/slip-0173.md

keepkeyjon, 2019-02-19:
@tynes which slip44 number should it have? Is matching testnet, ie `0x80000000|1` ok?

tynes, 2019-02-19:
Matching testnet is perfect, I believe that is the widely used coin type for regtest and its the coin type that we use regtest

https://github.com/satoshilabs/slips/blob/master/slip-0044.md
This link says that all testnets should use `1h`, which should include regtest

PR #87 (2019-02-27): Add Regtest CoinTable entry
Closes #86
commit 10ff88d7 (2019-02-27): Add Regtest CoinTable entry

Closes #86
closed 2019-02-27 by keepkeyjon