keepkey/keepkey-firmware #93: Coin integration

Jenova7, 2019-03-07:
Hi, i updated coins.def for Electra integration: https://github.com/Jenova7/keepkey-firmware/blob/2537cc4f2b484c2d3149341b8ff433ca3495590f/include/keepkey/firmware/coins.def#L24

Is it enough or what are the other necessary points?

Thanks

keepkeyjon, 2019-03-07:
You'll need to integrate https://github.com/keepkey/keepkey.js into a wallet that supports Electra, or do an Electrum port.

If you'd like to get Electra added to the new ShapeShift platform (coming soon), the best place to request it is the Asset Request Form: https://info.shapeshift.io/faq/#1529329581930-20752a4e-9e45

Jenova7, 2019-03-07:
thanks for the answer 👍

keepkeyjon, 2019-03-07:
Feel free to submit that coins.def change as a PR though.

keepkeyjon, 2019-03-09:
https://github.com/keepkey/keepkey-firmware/pull/94

closed 2019-03-09 by keepkeyjon

Jenova7, 2019-03-22:
`You'll need to integrate https://github.com/keepkey/keepkey.js into a wallet that supports Electra, or do an Electrum port.`

hi @keepkeyjon , we have now a electrumX server running: https://electrumx.electraproject.org/

keepkeyjon, 2019-03-22:
@Jenova7 you may want to merge from upstream to get this commit: https://github.com/spesmilo/electrum/commit/ef7af73bd3e14a81a53ddbbbe33526be79d03c2d