keepkey/keepkey-firmware #102: U2F webauthn acts flakey in chrome.

greatwolf, 2019-03-25:
I've reported this issue to the Chromium devs which you can find here: https://bugs.chromium.org/p/chromium/issues/detail?id=942454

Basically the issue is that for some sites that implement U2F it doesn't quite work for Keepkey. The device sees the registration/token request but after pressing the Keepkey button, Chrome does not detect this. What's strange is that if you press it the moment the request shows up on Keepkey then it goods. If you delay the press(eg. 1-2+ seconds later) then it doesn't work at all.

keepkeyjon, 2019-03-25:
Interesting, thanks for the report. What other sites have you seen this on besides the [yubico webauthn demo](https://demo.yubico.com/webauthn-technical/registration)?

I think our [upstream](https://github.com/trezor/trezor-mcu) has made some fido -> webauthn patches which we haven't pulled in yet... so most likely this is a problem on our end.

greatwolf, 2019-03-25:
I have tested this on a couple of sites so far.

Sites that work:

 - github.com
 - gitlab.com
 - bitfinex
 - twitter
 - gmail

Sites that are flakey:

 - mail.de
 - dropbox
 - bitbucket.org
 - fastmail.com

greatwolf, 2019-03-26:
Also wanted to add that facebook is interesting. I was able to register the token successfully but logging in and trying to authenticate, Keepkey actually displays the message `SW_COND_NOT_SAT`, a message I do not get on any of the other services.

keepkeyjon, 2019-03-26:
That's got to be this case:
https://github.com/keepkey/keepkey-firmware/blob/566920f8b4bcbaede73af36969e086a854d0b554/lib/firmware/u2f.c#L666-L674

keepkeyjon, 2019-03-26:
Is it supposed to return 'OK' in that case?

greatwolf, 2019-03-26:
I'm not familiar with U2F's protocol detail to tell if the implementation has the right behavior or not. I did notice however that sites that were working like github, gitlab they are also displaying the same `SW_COND_NOT_SAT` now on Chrome. I suspect Chromium devs might have changed something in v73.0.3683.86 since my reporting. I tried it on Opera which seems to use the same Blink engine as Chrome but an older version, that one works; no `SW_COND_NOT_SAT` is displayed (but it still exhibits the button delay problem for some sites like the older chrome version).

keepkeyjon, 2019-03-26:
> the recent Keepkey they got uses a different internal hardware(maybe different revision?)

All KeepKeys out in the wild currently have the same internal hardware (modulo small revisions, which wouldn't explain this difference in behavior).

> or they didn't initialize Keepkey with a 12-24 seed mnemonic. But for the latter case, is that even possible?

That would explain some of the behavior. If you try to do u2f things against an uninitialized device, it shows a message about it, and spits out SW_COND_NOT_SAT.

> Any idea why they're seeing such vastly different behavior than on mine?

Firmware-wise, I recommend using v6.0.0+ (preferably v6.0.4) for anything u2f. We had a v5.10.something that supported it, but had to pull that because of issues on windows, and have since fixed a couple of vulnerabilities in the u2f stack. New-in-box KeepKeys are going to have a pre v6.0.0 firmware on them, FWIW.

greatwolf, 2019-04-02:
Just wanted to add, a cursory look at the U2F spec suggest `SW_COND_NOT_SAT` is the right behavior. Turns out for sites where U2F is working they also show `SW_COND_NOT_SAT` but only for a split second(must be like <100ms). I just didn't notice it before because I wasn't looking at it closely enough.

I think what's actually happening is chrome is asking the token device if U2F is supported by sending that auth check. The device then returns `SW_COND_NOT_SAT` to indicate user interaction required which is how it indicates yes for U2F support. Chrome then sends the actually token auth request. But for some reason that second part never happens for some sites which explains why I just see `SW_COND_NOT_SAT` on my Keepkey.

I'm also curious whether others are having U2F issues on their Keepkey, I'll probably ask on /r/keepkey. Were you able to reproduce the problems I submitted in this issue?

keepkeyjon, 2019-04-02:
I haven't had a chance to try to repro it, but I'm sure it's a real issue on our end... I remember seeing similar flakiness on some of the u2f demo pages, but not on others.

I'll have to try one of them with a debugger hooked up and see what's really happening.

keepkeyjon, 2019-04-27:
I think what's wrong is that we're incorrectly following the spec. Registration / Authentication are supposed to be a two-step process, whenever user presence is required, and instead of replying immediately with SW_COND_NOT_SAT, we're just waiting for the user to confirm, and then sending the signature.

keepkeyjon, 2019-04-27:
Fixed here: https://github.com/keepkey/keepkey-firmware/commit/fe3e5e001c8143cd2fe60c22c2cdbf983b4a460c

Thanks so much for the report, @greatwolf!

greatwolf, 2019-04-29:
Thank you for looking into and fixing this. Does the commit fix the timing issues with the button confirmation?

btw, it looks like the recent update Chrome 74.0.3729.108 that `SW_COND_NOT_SAT` is no longer a problem but the button delay timing issue still persist.

Is there a test build of the firmware with the above commit available somewhere or do I need to build it from source? If I build it from source will there be any issues loading it onto the device since it isn't officially signed by Keepkey?

keepkeyjon, 2019-04-29:
Yeah, this fixes the timing issues.

You'd have to build from source, and if your bootloader is up to date, it won't run the unsigned firmware. We're shipping a new bootloader soon(ish) which will re-allow unsigned firmware, but that will most likely be after we've shipped this fix.

closed 2019-04-29 by keepkeyjon

keepkeyjon, 2019-04-30:
new bootloader: https://github.com/keepkey/keepkey-firmware/releases/tag/bl_v2.0.0
new firmware with the fix: https://github.com/keepkey/keepkey-firmware/releases/tag/v6.1.0

greatwolf, 2019-05-01:
I just loaded bootloader 2.0.0 and firmware 6.1.0 onto my keepkey. I can confirm that U2F is fully working now on Chrome. Unfortunately the Chrome Keepkey client no longer recognizes my Keepkey. The browser console for background.js says `Unknown device or version`.

keepkeyjon, 2019-05-01:
> Unfortunately the Chrome Keepkey client no longer recognizes my Keepkey. 

That's expected. I haven't shipped the client side update yet.

keepkeyjon, 2019-05-01:
>  I can confirm that U2F is fully working now on Chrome. 

Wonderful!

keepkeyjon, 2019-05-01:
I should have the new client out later this afternoon.

commit fe3e5e00 (2019-04-27): u2f: fix #102 'U2F webauthn acts flakey in chrome'

One thing of note in this patch: button animation behaves slightly differently
than the rest of the confirm dialogs, as the animation happens *afer* the
action instead of just before. This is necessary because the U2F spec demands
answers to requests before our animation would have finished. As a "good
enough" measure, instead we send the response at the moment when the user
starts pressing the button, instead of when they release at the end of the
animation.

Fixes #102