keepkey/keepkey-firmware #113: Can't apply to support page. Can't send tokens

NukeManDan, 2019-04-22:
I can not apply to the support : https://keepkey.zendesk.com/hc/en-us
`verification error`

I am unable to send any tokens - I get the full dialog on the keepkey w/ pin entry, but after confirming the tx, the chrome app hangs and nothing shows up on any block explorers. 

Please advise.

NukeManDan, 2019-04-22:
Using:
- Brave as base browser w/ chrome extension.
- Arch Linux

keepkeyjon, 2019-04-22:
Use the beta firmware. There's an issue with how the current prod firmware handles gas limits / gas fees.

keepkeyjon, 2019-04-22:
> I can not apply to the support : https://keepkey.zendesk.com/hc/en-us
> `verification error`

You can also email support: support@keepkey.zendesk.com

closed 2019-04-24 by keepkeyjon