keepkey/keepkey-firmware #134: Can't build KeepKey emulator (Ubuntu 18,  GCC 7)

gladcow, 2019-06-10:
I've failed to build KeepKey emulator.

What I'm doing:

`git clone https://github.com/keepkey/keepkey-firmware`
`cd keepkey-firmware`
`git submodule update --init --recursive`
`mkdir build`
`cd build`
`cmake -C ../cmake/caches/emulator.cmake ../ -DNANOPB_DIR=/path/to/nanopb-0.2.9.2 -DPROTOC_BINARY=/usr/bin/protoc`
`make`

Simple cmake output (command `cmake -C ../cmake/caches/emulator.cmake ../ -DNANOPB_DIR=home/sergey/src/nanopb-0.2.9.2 -DPROTOC_BINARY=/usr/bin/protoc >>cmake.out 2>&1`): https://pastebin.com/TAkT3nNa

CMake output in trace-expand mode (command `cmake --trace-expand -C ../cmake/caches/emulator.cmake ../ -DNANOPB_DIR=path/to/nanopb-0.2.9.2 -DPROTOC_BINARY=/usr/bin/protoc >>cmake.out 2>&1`): https://www.sendspace.com/file/gmxtve (sorry, it is too big, can't place it at pastebin)

make output (command `make >> make.out 2>&1`): https://pastebin.com/CijBjZDw

I'm not sure, but looks like `kkfirmware` lib with unresolved references is linked before `kkemulator` lib that contains implementations.

keepkeyjon, 2019-06-10:
Mind trying https://github.com/keepkey/keepkey-firmware/pull/135?

keepkeyjon, 2019-06-10:
It won't fix all the warnings from the missing forward declarations, but it should help out at link time.

gladcow, 2019-06-10:
Sorry to say, but it looks like it doesn't work,  still linking errors.( Output is https://pastebin.com/kynzphWD 

I'm still debugging it,  I can add an additional info after I'll find an exact reason.

keepkeyjon, 2019-06-10:
Looks like it never built `strlcpy.c`/`strncpy.c`. Maybe try blowing away your build directory & starting fresh? CMake caches a lot of stuff...

gladcow, 2019-06-10:
Sorry, I have misinformed you at the beginning.( strlcpy.c and strlcat.c were not added to sources really,  condition `if(NOT ${KK_HAVE_STRLCAT})` doesn't work,  may be because `${KK_HAVE_STRLCAT}` is undefined if we haven't strlcpy,  and so 'NOT ${KK_HAVE_STRLCAT}` is not defined too. 

I've tried
`if(${KK_HAVE_STRLCPY})
else()
    set(sources ${sources} strlcpy.c)
endif()

if(${KK_HAVE_STRLCAT})
else()
    set(sources ${sources} strlcat.c)
endif()`
and it is working correctly for me.

Btw,  it seams it is a bad idea to move this files to kkboard lib, because `crypto-unit` has reference to `strlcpy` in keepkey-firmware/unittests/firmware/nano.cpp:15, but it is not linked with kkboard lib

gladcow, 2019-06-11:
```
if(NOT DEFINED ${KK_HAVE_STRLCPY})
    set(sources ${sources} strlcpy.c)
endif()

if(NOT DEFINED ${KK_HAVE_STRLCAT})
    set(sources ${sources} strlcat.c)
endif()
```
is working for me too.

closed 2019-06-11 by keepkeyjon

gladcow, 2019-06-15:
Looks like it is CMake issue: https://gitlab.kitware.com/cmake/cmake/issues/19379
So,  the correct way to check this variables is something like `if(NOT KK_HAVE_STRLCPY)` and `if(NOT KK_HAVE_STRLCAT)`

keepkeyjon, 2019-06-17:
Oh wow, thanks for finding the root cause on that. Implemented in: https://github.com/keepkey/keepkey-firmware/commit/e2b64e097a0ed6952fd1fc70dd49eaef880b22aa

PR #135 (2019-06-11): move strlcpy/strlcat to libboard
Fixes #134
commit b16e5568 (2019-06-11): move strlcpy/strlcat to libboard

Fixes #134