keepkey/keepkey-firmware #165: Please Update Your Bootloader

aquiladev, 2019-09-02:
I try to setup a new keepkey. I have Windows. I installed Windows app for upgrading keepkey. Then upgraded bootloader and firmware by tutorial. Suddenly after upgrading firmware and reconnecting the keepkey I have an error "Please Update Your Bootloader".
I tried 3 times to pass tutorial, but I always have an issue with bootloader at the end

keepkeyjon, 2019-09-03:
> upgraded bootloader and firmware by tutorial

Which instructions are you following?

Can you show me the `Features` object for the device in bootloader mode? You can get this from python-keepkey, or go-keepkey by running either:

`keepkeyctl get_features`

or

`go-keepkey getFeatures`

after plugging the device in with the button held down.

aquiladev, 2019-09-04:
```
$ keepkeyctl get_features
No transport specified, attempting to detect device...
Device detected over HID
vendor: "keepkey.com"
major_version: 1
minor_version: 0
patch_version: 3
bootloader_mode: true
--

```

keepkeyjon, 2019-09-04:
That tells me your bootloader did not get updated by the tool. Should be `v1.1.0` or `v2.0.0`.

I recommend downloading the bootloader updater firmware from:

https://github.com/keepkey/keepkey-firmware/releases/download/bl_v1.1.0/blupdater.bin

Then upload that to your device with:

```
keepkeyctl -f blupdater.bin
```

Unplug and re-plug after it has been uploaded, and wait for that firmware image to update your bootloader (I believe this is the step that was missed/missing in the instructions). When it's finished, it will instruct you to unplug and enter bootloader mode again. This time, upload the latest firmware:

https://github.com/keepkey/keepkey-firmware/releases/download/v6.2.0/firmware.keepkey.bin

```
keepkeyctl -f firmware.keepkey.bin
```

Then you should be good to go.

aquiladev, 2019-09-05:
stuck on the first step
```
$ keepkeyctl -f blupdater.bin
usage: keepkeyctl [-h] [-v] [-t {usb,serial,pipe,socket,bridge,udp,webusb}]                                               
                  [-p PATH] [-Dt {usb,serial,pipe,socket,bridge,udp,webusb}]                                              
                  [-Dp DEBUGLINK_PATH] [-j] [-a] [--no-auto-button]                                                       
                  {apply_policy,change_pin,clear_session,decrypt_keyvalue,encrypt_keyvalue,eos_get_publ ic_key,eos_sign_tx,ethereum_get_address,ethereum_sign_tx,firmware_update,get_address,get_entropy,get_fe atures,get_public_node,list,list_coins,load_device,nano_get_address,nano_sign_tx,ping,recovery_device,r eset_device,set_label,sign_message,verify_message,wipe_device}                                                            

keepkeyctl: error: argument command: invalid choice: 'blupdater.bin' (choose from 'apply_policy', 'chan ge_pin', 'clear_session', 'decrypt_keyvalue', 'encrypt_keyvalue', 'eos_get_public_key', 'eos_sign_tx',  'ethereum_get_address', 'ethereum_sign_tx', 'firmware_update', 'get_address', 'get_entropy', 'get_featu res', 'get_public_node', 'list', 'list_coins', 'load_device', 'nano_get_address', 'nano_sign_tx', 'ping ', 'recovery_device', 'reset_device', 'set_label', 'sign_message', 'verify_message', 'wipe_device')
```

keepkeyjon, 2019-09-05:
oops, sorry, I meant:

```
keepkeyctl upload_firmware -f path/to/blupdater.bin
```

I'll edit my previous post to correct that.

keepkeyjon, 2019-09-05:
Alternatively, you can use the app, but when you get to the step after it uploads the bootloader updater, be sure to power cycle the device without holding the button, then follow the directions on the device's display.

closed 2019-09-10 by aquiladev