keepkey/keepkey-firmware #172: > Add HATCH.

hatchpay, 2019-09-26:
> Please force the addition of HATCH.

A few things:

1) Being pushy / impatient isn't going to get you what you want.
2) Asset request form is here: https://docs.google.com/forms/d/e/1FAIpQLScUlcAUrlShaM5QT7a9Mbj30WjBnN9femcz2iSVrtaqW4TsbA/viewform fill it out, and wait. Don't ask for progress on the request. We'll let you know if we decide to list HATCH.

_Originally posted by @keepkeyjon in https://github.com/keepkey/keepkey-firmware/pull/167#issuecomment-535522823_

We already filled out this form, we don’t have the opportunity to fill it out again, google does not allow it.

keepkeyjon, 2019-09-26:
> Don't ask for progress on the request. We'll let you know if we decide to list HATCH.

closed 2019-09-26 by keepkeyjon

hatchpay, 2019-09-26:
Are you talking about ShapeShift or KeepKey?

keepkeyjon, 2019-09-26:
Both

hatchpay, 2019-09-26:
> Both

Thank you. We look forward to hearing from you.