keepkey/keepkey-firmware #186: Support for more general output scripts

kyuupichan, 2019-10-30:
Keepkey does not support more general output scripts for signing, only basic templates like old-style limited OP_RETURN, P2SH and P2PKH.  Bitcoin's scripting language is far more flexible and more general support would be very useful.

keepkeyjon, 2019-10-30:
That added utility is a bit at odds with making the device simple for users to understand. I'd like to see a proposal for how to explain such scripts to a user in a way that doesn't decrease the security & ease of use of the existing setup. I have similar issues with https://github.com/keepkey/keepkey-firmware/issues/55

mrnerdhair, 2022-03-16:
For every TX we're asked to sign we have to have a way to tell the user what it does in plain language, and that's not something that can supported in general for arbitrary scripts.

closed 2022-03-16 by mrnerdhair