keepkey/keepkey-firmware #190: .text section overflows rom region

saagarjha, 2019-11-25:
I've been trying to build the firmware and I've been unable to successfully build a firmware that fits on my device. Building a release firmware gives me an error during the linking step both in the Docker image (with some munging to use the correct set of tools) and on my Mac with a manually-downloaded set of dependencies:

```
/opt/local/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: ../../bin/firmware.keepkey.elf section `.text' will not fit in region `rom'
/opt/local/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 66760 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [bin/firmware.keepkey.elf] Error 1
make[1]: *** [tools/firmware/CMakeFiles/firmware.keepkey.elf.dir/all] Error 2
make: *** [all] Error 2
```

A non-release build succeeds but does not install because it's "too large". Any ideas as to why this might be happening?

saagarjha, 2019-11-25:
```console
$ arm-none-eabi-nm --size-sort bin/firmware.keepkey.elf | tail
00003000 b msg.8271
00003000 B msg_resp.lto_priv.0
00003400 b decode_buffer.8255
00004000 b canvas_buffer
000040a4 t blake2b_compress
00005250 T tokens
00005350 T coins
00006000 T ge25519_niels_base_multiples
000090dc T nist256p1
000090dc T secp256k1
```

Commenting out https://github.com/keepkey/keepkey-firmware/blob/e1dcdc298adfcb1a8f555dd9e5407de459b09628/lib/firmware/ethereum_tokens.c#L10 fixed the problem. (Who needs Ethereum tokens anyways?)

keepkeyjon, 2019-11-25:
> A non-release build succeeds but does not install because it's "too large". Any ideas as to why this might be happening?

Known issue. We're getting close to the max size firmware that will fit on the device. We'll have to do some size optimizations before the next release.

I recommend starting from one of our tagged releases, and building with the scripts in `./scripts/build/docker/device/{release,debug}.sh`.

keepkeyjon, 2019-11-25:
also, re: removing ETH tokens: https://github.com/keepkey/keepkey-firmware/issues/68

saagarjha, 2019-11-26:
Thanks for the insight. Quick (but unrelated) question about building: the Docker images (v5-beta and v8) seem to default to the [docker-v6](https://github.com/keepkey/libopencm3/tree/docker-v6) branch of libopencm3, and I couldn't get it to build with any of the other branches even though the Dockerfile [seems to suggest otherwise](https://github.com/keepkey/keepkey-firmware/blob/5f9993aea31628832d56b4f34af12c62e59ae5b0/Dockerfile#L58). Is this expected?

keepkeyjon, 2019-11-26:
That's... strange. Good reminder to update that dependency though, since we haven't in a long time.

PR #200 (2019-12-16): firmware: actually build with -Os, not -O3
This brings the release build down to 492k from 630k.
    
Closes #190
commit 5439bdf0 (2019-12-16): firmware: actually build with -Os, not -O3

This brings the release build down to 492k from 630k.

Closes #190
closed 2019-12-16 by keepkeyjon