keepkey/keepkey-firmware #202: Stuck on "Preparing for upgrade"

clehner, 2019-12-21:
I used `keepkeyctl firmware_update` to install `blupdater.bin` from https://github.com/keepkey/keepkey-firmware/releases/download/v6.3.0/binaries.tar.bz2 after reading <https://github.com/keepkey/keepkey-firmware/issues/165#issuecomment-527941501>. The device now says Firmware Update Mode, Bootloader Version: 2.0.0, and the serial number. Now keepkeyctl from Ubuntu's python-keepkey 0.7.3-1 doesn't recognize the device (and I was not able to install the latest python-keepkey due to https://github.com/trezor/cython-hidapi/issues/72), and neither does the Keepkey Client Chrome Extension, but using go-keepkey works. But when I try to update the firmware to `firmware.keepkey.bin`, after confirming on the device, it shows Preparing for upgrade and an empty progress bar. How long is Preparing for upgrade supposed to take? Should I try wipeDevice?
```
$ go-keepkey getFeatures
Connected to 1 devices
  --
{
        "vendor": "keepkey.com",
        "major_version": 2,
        "minor_version": 0,
        "patch_version": 0,
        "bootloader_mode": true,
        "bootloader_hash": "dA9tSFC+jlVSbKvR7dYeg11FPkS/cjnxHd938oFrMTU=",
        "policies": [
                {
                        "policy_name": "bl_debug_link",
                        "enabled": false
                }
        ],
        "model": "K1-14AM",
        "firmware_hash": "im3CNoWv933AE45UGC9uXIAC0wjfqGsA6iLuFncMm/g="
}
BootloaderHash: 740f6d4850be8e55526cabd1edd61e835d453e44bf7239f11ddf77f2816b3135
```

keepkeyjon, 2019-12-22:
> How long is it supposed to take?

10s of seconds. You can interrupt it and try again.

> Should I try wipeDevice?

sure.

> Keepkey Client Chrome Extension does not recognize the device.

Try plugging in with the button held down, and then open the KeepKey client.

keepkeyjon, 2019-12-22:
You might have better luck with:

`go-keepkey upgradeFirmware`

or

`go-keepkey uploadFirmware`

keepkeyjon, 2019-12-22:
> Now when I try to update the firmware to firmware.keepkey.bin from v6.3.0

Another thought: did you pluck the unsigned binary out of the tarball? That might cause what you're seeing. You should use the signed one instead: https://github.com/keepkey/keepkey-firmware/releases/download/v6.3.0/firmware.keepkey.bin

clehner, 2019-12-22:
Attempting to upload https://github.com/keepkey/keepkey-firmware/releases/download/v6.3.0/firmware.keepkey.bin via keepkeyctl or go-keepkey still results in getting stuck at "Preparing for upgrade". go-keepkey upgradeFirmware is same. Keepkey Client recognizes the device (I think it was just a permissions issue before) but trying to upgrade the firmware from there does not make progress either. wipeDevice did not help. The device shows "FIRMWARE UPDATE REQUIRED" when plugged in with or without holding the button.
Is there something I could do with flashWrite to restore access?

keepkeyjon, 2019-12-22:
> The device shows "FIRMWARE UPDATE REQUIRED" when plugged in with or without holding the button.

This means that the device is in bootloader mode, and waiting for a firmware to be uploaded.

Not sure what's going on with your device. I recommend contacting support & seeing if they'll send you a replacement. Mention this thread in the support ticket.

closed 2019-12-22 by keepkeyjon

clehner, 2019-12-27:
Support asks me to run the following proprietary application:
https://beta.shapeshift.com/keepkey-updater-mac.dmg
via https://beta.shapeshift.com/updater-download
It requires macOS 10.13, which I don't have. They did not acknowledge my mentioning this thread.

keepkeyjon, 2019-12-27:
The updater app isn't going to help you, even if you do manage to get it installed... it hasn't been taught how to talk to the v2.0.0 bootloader yet.

> They did not acknowledge my mentioning this thread.

That's very disappointing. Not sure how else to help you :(

keepkeyjon, 2019-12-27:
How much of the progress bar is filled in when it gets stuck on "Preparing for Upgrade"? Mind attaching a photo of it?

clehner, 2019-12-30:
They are sending me a replacement device.
Without understanding of why this problem is happening, I am a little wary of that it could reoccur.
Maybe it had to do with that I did not initialize the device before trying to upgrade the firmware and bootloader?
Picture: [image]
From looking in the source code I would guess one of these functions in `handler_erase`
in `tools/bootloader/usb_flash.c` is blocking.

keepkeyjon, 2019-12-31:
I recommend not upgrading to the `v2.0.0` bootloader next time. Most devices out in the wild are on `v1.1.0`.