keepkey/keepkey-firmware #216: Emulator build

antonio-fr, 2020-01-18:
Can't make the emulator build at all in Ubuntu18.04 :

First the[ doc build ](https://github.com/keepkey/keepkey-firmware/blob/v6.3.0/docs/Build.md) is so outdated and so useless. This now requires a recursive git clone, and the given build directory fails.

```
cd ~
wget https://jpa.kapsi.fi/nanopb/download/nanopb-0.2.9.2.tar.gz
tar xzf nanopb-0.2.9.2.tar.gz
cd nanopb/generator/proto
make
export PATH=~/nanopb/generator:$PATH
cd ~
git clone https://github.com/keepkey/keepkey-firmware
cd keepkey-firmware
git checkout v6.3.0
git submodule update --init --recursive
mkdir build
cd build
cmake -C ../cmake/caches/emulator.cmake ../
make
```

I get the error message :
```
[ 98%] Building CXX object unittests/firmware/CMakeFiles/firmware-unit.dir/storage.cpp.o
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp: In member function ‘virtual void Storage_WriteMeta_Test::TestBody()’:
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp:38:5: error: C99 designator ‘uuid’ outside aggregate initializer
     };
     ^
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp:38:5: error: C99 designator ‘uuid_str’ outside aggregate initializer
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp: In member function ‘virtual void Storage_WriteCacheV1_Test::TestBody()’:
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp:188:5: error: C99 designator ‘root_seed_cache’ outside aggregate initializer
     };
     ^
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp: In member function ‘virtual void Storage_DumpNode_Test::TestBody()’:
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp:229:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp: In member function ‘virtual void Storage_UpgradePolicies_Test::TestBody()’:
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp:589:5: error: C99 designator ‘policy_name’ outside aggregate initializer
     };
     ^
/home/moa/keepkey-firmware/unittests/firmware/storage.cpp:589:5: sorry, unimplemented: non-trivial designated initializers not supported
unittests/firmware/CMakeFiles/firmware-unit.dir/build.make:206: recipe for target 'unittests/firmware/CMakeFiles/firmware-unit.dir/storage.cpp.o' failed
make[2]: *** [unittests/firmware/CMakeFiles/firmware-unit.dir/storage.cpp.o] Error 1
CMakeFiles/Makefile2:1477: recipe for target 'unittests/firmware/CMakeFiles/firmware-unit.dir/all' failed
make[1]: *** [unittests/firmware/CMakeFiles/firmware-unit.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

If I checkout the latest master and use nanopb 0.3.9.4, I have strictly the same error message.

Building with the docker script works. But then, it seems x86 and not AMD64, so difficult to use. Also @keepkeyjon is used to build it outside the script.

Any advice to compile and run emulator on Linux is welcome. I need to check because I can see some images artifacts on the keepkey and I investigate if this is a hardware issue or an issue in draw_bitmap_mono_rle.

keepkeyjon, 2020-01-20:
> I need to check because I can see some images artifacts on the keepkey and I investigate if this is a hardware issue or an issue in draw_bitmap_mono_rle.

screenshot?

keepkeyjon, 2020-01-20:
> I get the error message :

what compiler?

antonio-fr, 2020-01-20:
The issue is present with gcc version 8.3.0.
It compiles fine with clang version 7.0.1. Still, there are lots of warning regarding "implicit declaration of function 'strlcat'/'strlcpy' is invalid in C99".

antonio-fr, 2020-01-20:
"I can see some images artifacts on the keepkey"
Turns out there was two hardware effects combined.
1) The display is limited to 16 grey levels. I think the emulator renders all 256 grey tones according to the code. I will see that point and eventually patch it so the emulator renders the same.
2) The OLED screen renders scanlines differently whether the lines have lots of whites or lot of dark. e.g. a single white dot on a black line is lighter than the same white pixel on a white line.

keepkeyjon, 2020-01-22:
Besides the [Giant KeepKey](https://www.reddit.com/r/keepkey/comments/cgkq8x/our_senior_engineer_is_working_on_a_giant_keepkey/), there isn't a graphical frontend for the emulator. Do you have patches that add one?

> I think the emulator renders all 256 grey tones

Yeah, it does, and I don't think there's much we can do about (2) in the general case.

antonio-fr, 2020-01-24:
Oh I thought it was graphical. This is indeeed just the processing with a socket interface.
For the emulator, it would be nearer the reality if it renders (pixcol//16)*16 instead. Anyway this is not very important because there's no GUI.

About the "scanline" issue, this is a 100% hardware issue. Maybe this can be improved on the next hardware revision with better capacitors. Maybe this is the best the screen can deliver. Definitively not a firmware issue.

For this current issue about the compilation, it mainly needs to update the docs to be explicit about clang compiler is required.

keepkeyjon, 2020-01-27:
> For this current issue about the compilation, it mainly needs to update the docs to be explicit about clang compiler is required.

Is this still an issue after https://github.com/keepkey/keepkey-firmware/pull/217?

keepkeyjon, 2020-02-04:
I believe this has been fixed in https://github.com/keepkey/keepkey-firmware/commit/22c3b5cc89a990d9be7d5b44cf88882ee23362fa and https://github.com/keepkey/keepkey-firmware/commit/2b33e574111808d317d36b44bf954d04770258ec. Feel free to re-open if you're still having issues related to this.

closed 2020-02-04 by keepkeyjon