keepkey/keepkey-firmware #233: Does the current KeepKey FW support FIDO U2F usage on Chrome?

brianddk, 2020-04-18:
I see that call-outs in the Firmware, but I see some components have been removed in the library.  Can I use a KeepKey (on this FW rev) as a U2F device on `github.com` like I can with Ledger and Trezor?

My1, 2020-07-18:
at least on the latest firnware that is released U2F works. no idea whether this revision on github is any different

brianddk, 2020-07-31:
@My1 , Thanks... 

Closing

closed 2020-07-31 by brianddk

My1, 2020-08-01:
no problem. and in fact the Keepkey's u2f implementation is done very interestingly.
it uses the first 44 bits of the appid SHA-256 hash (it doesnt get the appid verbatim sadly) to create a mnemnoic you can use to make sure you are signing in to the correct place which is kinda awesome