keepkey/keepkey-firmware #235: Missing compiled Firmware Binary for v6.5.1 in the Release Download section.

greatwolf, 2020-10-10:
See Title. I'm of course talking about this link: https://github.com/keepkey/keepkey-firmware/releases. Where's the corresponding Changelog for this version?

Also, the commit bumps the bootloader to version 2.1.0 up from 2.0.0. Again where is the binary for that new bootloader?

greatwolf, 2020-10-19:
As I've posted on reddit [here](https://www.reddit.com/r/keepkey/comments/jbq6hu/keepkey_firmware_update_651/g91x8dr?utm_source=share&utm_medium=web2x&context=3), I've the link to the latest firmware.keepkey.bin.

The relevant urls:

 - Table of Content list of versions: https://static.shapeshift.com/firmware/releases.json
 - Latest binary(v6.5.1 at time of this posting): https://static.shapeshift.com/firmware/v6.5.1/firmware.keepkey.bin

With @keepkeyjon no longer here it seems, who's responsible for maintaining this firmware repository? When is the release section going to be updated with the latest signed binary build?

greatwolf, 2020-10-22:
Closing this issue since it's been resolved.

closed 2020-10-22 by greatwolf

reidrankin, 2021-07-09:
Sorry for the confusion regarding this; our processes around firmware release have been an  "area of potential improvement" for a while, and we're working to coordinate binary releases and source code/changelog updates better.

In answer to your question, there is no bootloader v2.1.0 -- or, at least, there's not an official version with that number yet. Our intention at the moment is to use that version number for signing key rotation in the future; bootloader v2.0.0 added support for the notion of signatures with "expired keys", but we haven't actually used that functionality yet.

(I say there's no _official_ version, because the build scripts build the bootloader automatically each time they're run. Since we bumped the version number, each time you build you're also going to produce an unsigned bootloader that's identical to 2.0.0 except it's numbered 2.1.0.)