keepkey/keepkey-firmware #281: Manual building of latest 5ab76aa results in a broken firmware binary.

greatwolf, 2021-07-10:
I'm running into an issue where building the latest v7.1.5 5ab76aa results in a non-functional firmware. I'm able to upload onto my keepkey fine but afterwards my keepkey would be stuck on startup becoming non-responsive. The animated KeepKey logo never shows up nor does the `get/started` welcome screen.

The warning of unofficial firmware does come up however. After confirming that is where my keepkey just stays in a non-responsive state.

Steps to reproduce:
 - git clone to the latest tip of the repo
 - recursively init and update all the dependent submodules as needed for current tip
 - cd into `./scripts/build/docker/device`
 - run `./release.sh` build script
 - after build completes upload the the produced firmware binary file `./bin/firmware.keepkey.bin` onto keepkey device.
 
I've tried diagnosing the issue and so far I've narrowed it down to some commit introducing this between bc40158 and 852f8ac.
Commit 852f8ac works fine btw, I am able to produce a working firmware from that.

greatwolf, 2021-07-11:
After some more investigation and debugging I've narrowed down the problem to commit 0527170. Specifically the problematic [line is](https://github.com/keepkey/keepkey-firmware/commit/05271703b8c1863262f6df0d089d273fc2749ac8#diff-0b10186cddda4783b1e93def8cdd72dd91e0e42b4cc411fbfacf1d09c1fb7be8R97):

    return !fi_defense_delay(sigPresent);

In the original line before this commit you have:

    return *((uint8_t *)FLASH_META_SIGINDEX1) != 0;

I haven't looked inside `fi_defense_delay` but assuming that function just passes through `sigPresent` as a return or converts it to a `bool` then to keep the semantics of the original piece of code it should be something like this instead:

    return fi_defense_delay(sigPresent) != 0;
    // or simply
    return fi_defense_delay(sigPresent);

Changing that line and recompiling produces a working firmware again. My KeepKey no longer seizes up and becomes unresponsive.

markrypto, 2021-07-12:
greatwolf, what bootloader are you using?

reidrankin, 2021-07-12:
Thanks for bringing this to our attention. We didn't catch this in testing because it doesn't occur on DEBUG_LINK devices, which is what we usually test builds on before they're signed. We've pulled the broken binary and will be releasing a fixed one as v7.1.6 later today or early tomorrow.

closed 2021-07-13 by markrypto

reidrankin, 2021-07-13:
@greatwolf This is fixed in [v7.1.7](https://github.com/keepkey/keepkey-firmware/releases/tag/v7.1.7).

greatwolf, 2021-07-13:
> greatwolf, what bootloader are you using?

This is on bootloader v2.0.0