keepkey/keepkey-firmware #283: Reproducing v6.2.2 - v7.1.5 requires a case-insensitive FS

reidrankin, 2021-07-12:
There's a build reproducability issue with the KK firmware. Between `v6.2.0` and `v6.2.2` a commit was made to the `keepkey/ethereum-lists` repo that duplicated the JSON metadata file for the PAXG token. The original file was named `0x45804880De22913dAFE09f4980848ECE6EcbAf78.json`; a copy was made at `0x45804880de22913dafe09f4980848ece6ecbaf78.json` (note the case change). Our builds have all been done manually by engineers on Mac OS, which uses case-insensitive FS by default, and so only actually checks out one of the identical files.
The fix is fairly simple: update the `ethereum-lists` repo to lowercase all the filenames. That said, it will require a submodule update on `python-keepkey` and again on `keepkey-firmware`, which will change the commit hash embedded in the FW binary.

reidrankin, 2021-07-12:
This will required keepkey/ethereum-lists#1 and keepkey/python-keepkey#122 to land "upstream"

Giszmo, 2021-07-19:
I'm pretty sure I followed the build instructions and failed to reproduce v7.1.7, too.

```
$ sha256sum firmware.keepkey.bin ; tail -c +257 firmware.keepkey.bin | sha256sum ; tail -c +257 ./bin/firmware.keepkey.bin | sha256sum
2b7edd319536076e0a00058d0cfd1b1863c8d616ba5851668796d04966df8594  firmware.keepkey.bin
5528034fec8334a7ee494c2ec50d5f0368e2e5fe403f4bc29a54c70fa026e2c0  -
14b831edbe0555dd1f1c11f98cf1a42338b4058274496b7f45f0f66d9523ff94  -
```

Full documentation on [WalletScrutiny](https://walletscrutiny.com/hardware/keepKey/).

reidrankin, 2021-07-19:
We released v7.1.7 as a hotfix on top of v7.1.5, so it doesn't have this issue fixed yet. I'll updated the title of this issue to reflect that.

You should be able to reproduce the binary even on a case-sensitive filesystem by manually removing the copy of the PAXG token metadata with the mixed-case filename. Let me know if you still have trouble after that; if you've found another build reproducibility issue that we're unaware of, we definitely want to get that fixed.

Giszmo, 2021-07-19:
Sorry, which file do I have to delete at which step? Is it after `git submodule update --init --recursive`? What's the exact path?

reidrankin, 2021-07-28:
(@Giszmo Very sorry this reply took so long; entirely my fault. I've had this reply sitting open in my browser on the "preview" tab for more than a week, without realizing I never mased the big green button!)

Here's a set of commands that should allow you to reproduce the v7.1.7 build; I've just tested these myself on Ubuntu 21.04.

```
git clone https://github.com/keepkey/keepkey-firmware
cd keepkey-firmware
git checkout v7.1.7
git submodule update --init --recursive
rm deps/python-keepkey/keepkeylib/eth/ethereum-lists/src/tokens/eth/0x45804880de22913dafe09f4980848ece6ecbaf78.json
./scripts/build/docker/device/release.sh
tail -c +257 bin/firmware.keepkey.bin | sha256sum | cut -d ' ' -f 1
```

I'm also working on a comprehensive script for automatic build reproducibility verification; still fighting a few issues getting it running reliably on Mac OS, but I'll make sure to post it up here when done.

Giszmo, 2021-07-31:
Thanks @reidrankin ! Can confirm your instructions work to reproduce 7.1.7. [WalletScrutiny updated accordingly](https://walletscrutiny.com/hardware/keepKey/#upd0731).

PR #292 (2021-08-03): Fix build repro issue
Fixes #283. Requires https://github.com/keepkey/ethereum-lists/pull/2 and https://github.com/keepkey/python-keepkey/pull/122 to land first.
closed 2021-08-03 by markrypto