keepkey/keepkey-firmware #305: Add support for Binance Smart Chain transactions (MyEtherWallet)

DJFliX, 2021-12-25:
Hi! I wanted to transfer funds over the Binance Smart Chain using my KeepKey with MyEtherWallet. Unfortunately I have discovered that while MEW is able to see funds associated with my Ethereum address on the Binance Smart Chain it is not possible to sign a transaction with the KeepKey. 

```
Incompatible EIP155-based V 147 and chain id 1. See the Common parameter of the Transaction constructor to set the chain id.
```

[image]
 
Is support for the Binance Smart Chain achievable for the KeepKey? I see tons of open PR's and no real presence from keepkey staff so I'm quite hesitant to spend time on any form of implementation unless there is some certainty that it will actually be merged (if it meets the code quality standards obviously).

BitHighlander, 2022-02-11:
Hey @DJFliX
     Interesting, I didn't realize BSC tx format would need a special patch. I know onboard.js supports BSC so this would be useful. We would welcome a pr patching this. perhaps I'll take a look at getting some BSC reference TX's into https://github.com/shapeshift/hdwallet and we could backport into keepkey. 

Thanks for posting

mrnerdhair, 2022-03-16:
Context: https://www.reddit.com/r/MyEtherWallet/comments/rllvmw/transaction_on_bsc_chain_fails_incompatible/

mrnerdhair, 2022-03-16:
The KK's firmware [supports](https://github.com/keepkey/keepkey-firmware/blob/master/lib/firmware/ethereum.c#L627-L638) alternate chain IDs natively. I've been looking through the MEW code trying to figure out where this is coming from. I've found some similar issues (like [this](https://github.com/NomicFoundation/hardhat/blob/e822c45da49fa1717d42c4ba9993644e7550b2bc/docs/metamask-issue.md) and [this](https://github.com/trufflesuite/ganache/pull/532)), and I don't think the problem is with KK -- the error strings appear to be from the Ethereum RPC server and are indicating that the KK is signing with a chain ID of 147 (which it would do if told to, but that would be really strange given that BSC's chain ID is 56) but something's expecting a chain id of 1 (which is also definitely wrong for BSC).

I don't know what MEW's backend looks like, but maybe they have something misconfigured somewhere?

DJFliX, 2022-03-17:
@mrnerdhair Thanks for chiming in! I've quoted your latest response in the reddit thread and I've asked if there is a way for me to provide debug logs so that someone from MEW can verify whether the interaction between MEW and the KeepKey is working as they expect. If there are further developments I'll mention them here as well.

BitHighlander, 2023-04-27:
BSC is now fully supported in app.shapeshift.com. MEW issues should be issued on their repo.

closed 2023-04-27 by BitHighlander

DJFliX, 2023-04-27:
Amazing, thanks a lot. I'll seek out my KeepKey and try it out soon!