keepkey/keepkey-firmware #319: Questions on Current Security Reporting and Bug Bounty Policies

invd, 2022-08-08:
Hello @pastaghost and @markrypt0, 
in early 2022, the ownership of KeepKey changed in complex ways (as outlined [in the public SCP-71 proposal](https://forum.shapeshift.com/t/scp-71-proposal-to-create-the-keepkey-workstream)) and I suspect that the [SECURITY.md](https://github.com/keepkey/keepkey-firmware/blob/master/SECURITY.md) no longer correctly represents the project ownership structure, which may also affect the security contact and bug bounty policy details.

Could you confirm that the current policies documented in `SECURITY.md` and on https://shapeshift.com/responsible-disclosure-program are still accurate - in other words, that the ShapeShift DAO operates a bug bounty program for `keepkey-firmware` and that you want vulnerability disclosures to get reported to `security@shapeshift.io`?
Thanks!

markrypt0, 2022-08-08:
Good question. I believe the bug bounty program is still in place but this doc may need some updates as far as where to report, @pastaghost can confirm.

invd, 2022-12-29:
The https://shapeshift.com/responsible-disclosure-program has been updated and now lists KeepKey under the out-of-scope section via `Physical (or emulated!) KeepKey devices (KeepKey is now a separate company)` .

@pastaghost and @markrypt0: Is there a followup bug bounty program in place by KeepKey? Can you update the outdated `SECURITY.md` to describe the current status?

In any case, please provide a contact email addresses and optionally PGP key(s) for new issue submissions.

markrypt0, 2022-12-29:
I will bring this up again with the owners, thank you for pointing this out.

markrypt0, 2022-12-29:
Progress has been made but not updated yet. It may take several days still, due to the holidays.

invd, 2023-01-16:
@markrypt0 : I've noticed the new [SECURITY.md](https://github.com/keepkey/keepkey-firmware/blob/f946abf7cb089564a6a0de649a9d310179215135/SECURITY.md) content which points to https://www.keepkey.com/security .

This is a significant improvement since the new documentation clarifies where to report issues to and provides a new PGP pubkey. However, as far as I can tell neither the `SECURITY.md` nor https://www.keepkey.com/security specifies if there is still a bug bounty program in place, and what its conditions are. Can you clarify the new status of this?

Also, I think there are some formatting issues in both places. For the Markdown file, this should be easy to fix by closing the ```  section at the bottom.

markrypt0, 2023-01-16:
Thanks for the formatting catch. As far as the bug bounty, I would suggest emailing via the address in the link and asking them directly.

closed 2023-01-16 by markrypt0

invd, 2025-05-03:
@markrypt0 quick note that the mentioned `SECURITY.md` formatting is still broken. Additionally, neither the markdown file nor the website it links to actually specify _which email address_ should be used to report vulnerabilities. The PGP key has `security@keepkey.com` in its metadata, but this is not immediately obvious to potential submitters.