keepkey/keepkey-firmware #342: v7.8.0 is not reproducible

mohammadrafigh, 2023-05-22:
I followed the build process but received 2 different hashes from the binary in repo and what I built:

github: `b0fc047c7789bee9fba72e5720fb5c8129c9a32f94f7fe9dca4f4675c8d3ddc2`
built binary: `aa5d303ae15d2b5dd3ace06ad4d8665c644efbfba1946c25aa5e25d6d17ed917`

Giszmo, 2023-07-13:
For more context, our full review can be found [here](https://walletscrutiny.com/hardware/keepKey/).

We tested 7.8.0 with hash 31c1cdd945a7331e01b3cced866cb28add5b49eef87c2bbc08370e5aa7daf9bf

Our test script:

```
#!/bin/bash

### provide this script with the version without "v" and the published buildHash

version=$1
buildHash=$2

cd /tmp
git clone https://github.com/keepkey/keepkey-firmware
cd keepkey-firmware
git checkout v${version}
git submodule update --init --recursive
# rm deps/python-keepkey/keepkeylib/eth/ethereum-lists/src/tokens/eth/0x45804880de22913dafe09f4980848ece6ecbaf78.json
./scripts/build/docker/device/release.sh
wget https://github.com/keepkey/keepkey-firmware/releases/download/v${version}/firmware.keepkey.bin
sha256sum firmware.keepkey.bin
tail -c +257 firmware.keepkey.bin | sha256sum
tail -c +257 ./bin/firmware.keepkey.bin | sha256sum
```

eminentnitesh, 2025-10-05:
how to firmware update in keep key